PubInv/PMD_-pseudo-medical-device-aka-Homework2- #1: The Copper Fill is Not Connected

ForrestErickson, 2024-04-17:
## Summary
As of 20240417 the front copper fill is not connected to a net.
[image]

### Suggested Fix
It would be best to connect to the net "GND".
Then back in the PCB editor, press the "b" to repour the fill. 

Show the results in another comment.

nk25719, 2024-04-20:
[image]

I followed your suggestions and this is the result image.

ForrestErickson, 2024-04-20:
Lee confirms the change has fixed the issue. The copper fill is connected to GND.

closed 2024-04-20 by ForrestErickson