PubInv/PMD_-pseudo-medical-device-aka-Homework2- #2: Add Net Classes To Schematic and PCB

ForrestErickson, 2024-04-20:
## Summary:
The KiCad Net classes help us manage trace width, spacing and Via diameters in a PCB layout.

## References 
This reference may help:
https://forum.kicad.info/t/kicad-7-0-1-netclasses-in-schematic-editor-totally-lost-noob/41275

See also for temperature rise the concepts here at JLCPCB:
https://jlcpcb.store/blogs/blog-post/jlcpcb-x-easyeda-pcb-trace-width-and-clearance-calculation

## TO DO
Decide on the maximum ambient temprature on which the GPAD will opperate

ForrestErickson, 2024-04-20:
Make the GND net class a minimum with of 40 mills and a maximum of 100 mills
Make the signal trace default trace width 20 mills and minimum of 10 mills.
Make power trace width 30 mills default and 30 mills minimum.

Each of these will be a "net class" which may or may not be a "directive".

ForrestErickson, 2024-04-20:
@nk25719 

Although I have not watched all of it, this video title looks promising:
https://www.youtube.com/watch?v=Fv9zQ3nkLqk

nk25719, 2024-04-28:
The Net Classes are added in the board setup. 

[image]  

>> This screenshot shows the Netclasses and the connections assigned to each pattern.

[image]  

>> This screenshot shows an example of how the width of the trace used to connect the +5V is calculated, taking into consideration a temperature rise of 40 degrees Celsius.

nk25719, 2024-04-28:
@ForrestErickson 
The updated PCB copper routing with the added Net Classes is on the <sup>Net-Classes</sup> branch.

Before closing this issue and merging the <sup>Net-Classes</sup> branch with the <sup>main </sup> branch.

- [x] Add Net Classes to PCB 
- [ ] Add Net Classes to schematic
- [ ] Use JLCPCB rules

nk25719, 2024-04-28:
> [!NOTE]
> The JLCPCB rules will effect the net classes and we might need to edit the added classes to comply with the rules of minmum width trace width and spacing required. The PCB should be rerouted after editing the sizes of the vias, holes and NTPH.  

> [!TIP]
> PCB Manufacturing & Assembly Capabilities: https://jlcpcb.com/capabilities/pcb-capabilities

> [!IMPORTANT]
>The minimum NPTH dimension is 0.50mm, Please add the NPTH in the mechanical layer or keep out layer.

> [!CAUTION]
> Advise: before starting with a PCB check and note down the rules of JLCPCB. 

> [!WARNING]
> Make sure to make a pull of the Net-Classes branch and not the main branch.

nk25719, 2024-05-04:
I believe the Net Classes are updated automatically on the Schematic Editor when we update the Net Classes on the PCB Editor. 

[image]

> Screenshot of the Schematic Setup window.

nk25719, 2024-05-04:
(rule "Track width, outer layer (1oz copper)"
	(layer outer)
	(condition "A.Type == 'track'")
	(constraint track_width (min 0.127mm))
)

(rule "Track spacing, outer layer (1oz copper)"
	(layer outer)
	(condition "A.Type == 'track' && B.Type == A.Type")
	(constraint clearance (min 0.127mm))
)

(rule "Track width, inner layer"
	(layer inner)
	(condition "A.Type == 'track'")
	(constraint track_width (min 0.09mm))
)

(rule "Silkscreen text"
	(layer "?.Silkscreen")
	(condition "A.Type == 'Text' || A.Type == 'Text Box'")
	(constraint text_thickness (min 0.15mm))
	(constraint text_height (min 1mm))
)

(rule "Pad to Silkscreen"
	(layer outer)
	(condition "A.Type == 'pad' && B.Layer == '?.Silkscreen'")
	(constraint silk_clearance (min 0.15mm))
)

(rule "Edge (routed) to track clearance"
	(condition "A.Type == 'track'")
	(constraint edge_clearance (min 0.3mm))
)

ForrestErickson, 2024-05-08:
@nk25719 

Regarding, "I believe the Net Classes are updated automatically on the Schematic Editor when we update the Net Classes on the PCB Editor."

Did you mean the following:
"I believe the Net Classes become available in the Schematic Editor and are update when we create Net Classes on the PCB Editor."

Or something else?

nk25719, 2024-05-19:
As soon as we close this issue we should be merging the Net-Classes branch with the main branch.

nk25719, 2024-06-11:
We reviewed this issue with Lee, and we have successfully added net classes for the schematic and the printed circuit board. We are closing this issue.

closed 2024-06-11 by nk25719