PubInv/PMD_-pseudo-medical-device-aka-Homework2- #3: PCB Deign Should be JLCPCB Compiant, IE, Use JLCPCB Design Rules

ForrestErickson, 2024-04-23:
# Summary
Every PCB Manufacturer has it's own capabilities and some may not be able to manufacture PCBs that other manurfacturiers can.
Design for low cost means design that does not require highest precision of the manufacturiers. Such design also makes for lower reject rate on PCBs due to manufacturing defects.

## Starting Somewhere
To get started lets review the requirements for JLCPCB manufacture of PCBs.
Their published requirements are found here:  https://jlcpcb.com/capabilities/pcb-capabilities  

Some of these requirement  map directly into the KiCad PCB editor setup but not all.

## A Possible Short Cut
This repository may be all we need, but I have not yet tried.
https://github.com/labtroll/KiCad-DesignRules
Review of what this author did may be of help, but we may still want to tweek for our own reasons.

ForrestErickson, 2024-04-23:
Here is the settings as we were working via Discord on 20240423 (and lee was tired too.)
[image]

nk25719, 2024-05-04:
JLCPCB Rules I followed:

(rule "Track width, outer layer (1oz copper)"
(layer outer)
(condition "A.Type == 'track'")
(constraint track_width (min 0.127mm))
)

(rule "Track spacing, outer layer (1oz copper)"
(layer outer)
(condition "A.Type == 'track' && B.Type == A.Type")
(constraint clearance (min 0.127mm))
)

(rule "Track width, inner layer"
(layer inner)
(condition "A.Type == 'track'")
(constraint track_width (min 0.09mm))
)

(rule "Silkscreen text"
(layer "?.Silkscreen")
(condition "A.Type == 'Text' || A.Type == 'Text Box'")
(constraint text_thickness (min 0.15mm))
(constraint text_height (min 1mm))
)

(rule "Pad to Silkscreen"
(layer outer)
(condition "A.Type == 'pad' && B.Layer == '?.Silkscreen'")
(constraint silk_clearance (min 0.15mm))
)

(rule "Edge (routed) to track clearance"
(condition "A.Type == 'track'")
(constraint edge_clearance (min 0.3mm))
)

(rule "Hole diameter"
(constraint hole_size (min 0.2mm) (max 6.3mm))
)

(rule "Hole (NPTH) diameter"
(layer outer)
(condition "!A.isPlated()")
(constraint hole_size (min 0.5mm))
)

(rule "Hole (castellated) diameter"
(layer outer)
(condition "A.Type == 'pad' && A.Fabrication_Property == 'Castellated pad'")
(constraint hole_size (min 0.6mm))
)

(rule "Annular ring width (via and PTH)"
(layer outer)
(condition "A.isPlated()")
(constraint annular_width (min 0.075mm))
)

(rule "Clearance: hole to hole (perimeter), different nets"
(layer outer)
(condition "A.Net != B.Net")
(constraint hole_to_hole (min 0.5mm))
)

(rule "Clearance: hole to hole (perimeter), same net"
(layer outer)
(condition "A.Net == B.Net")
(constraint hole_to_hole (min 0.254mm))
)

(rule "Clearance: track to NPTH hole (perimeter)"
(condition "!A.isPlated() && B.Type == 'track' && A.Net != B.Net")
(constraint hole_clearance (min 0.254mm))
)

(rule "Clearance: track to PTH hole perimeter"
(condition "A.isPlated() && B.Type == 'track' && A.Net != B.Net")
(constraint hole_clearance (min 0.33mm))
)

(rule "Clearance: track to pad"
(condition "A.Type == 'pad' && B.Type == 'track' && A.Net != B.Net")
(constraint clearance (min 0.2mm))
)

(rule "Clearance: pad/via to pad/via"
(layer outer)
(condition "A.isPlated() && B.isPlated() && A.Net != B.Net")
(constraint clearance (min 0.127mm))
)

ForrestErickson, 2024-05-08:
@nk25719 
For example regarding, "(rule "Hole diameter" (constraint hole_size (min 0.2mm) (max 6.3mm)) )"

We should discuss what is our goal with respect to JCLCPB design rules.

First: We should set up KiCad so that we do not violate the JCLCPB design rules.
However, we may want a KiCad setup that has even more generous (more conservative) design rules that make manufacture perhaps more portable.
There is a way to import design rules and so I infer a way to export and capture a design rules setup.
Perhaps we should be capturing and naming these two (or more) design rule sets.

nk25719, 2024-05-21:
https://jlcpcb.com/capabilities/pcb-capabilities

nk25719, 2024-05-21:
By updating the design rules to reflect the jlcpcb requirements. 
Changed the via size to meet the JLCPCB defined sizes, and we added Via predefined sizes and track predefined sizes.
DRC rule check shows zero Errors.

[image]

[image]

nk25719, 2024-06-11:
We edited the Design rules contrains to the JLCPCB requirements. Lee and I revied and closed this issue.

closed 2024-06-11 by nk25719