PubInv/PMD_-pseudo-medical-device-aka-Homework2- #4: The Power Jack (AKA Barrel Jack) Needs To Be Near / Overhang Board Edge.

ForrestErickson, 2024-04-23:
#  Summary
The layout of 20240421 has poorly placed Power Jack. 
The PCB will interfer with fingers 
[image]

## Suggestion(s)
The barrel jack must overhang the edge of the PCB.  So that it might be flush with an enclosure.
In which Direction / edge of the PCB should the power jack be located?

nk25719, 2024-04-27:
I received error messages when I let the barrel jack overlay the courtyard, and also the silkscreen was outside the Front silkscreen layer. 
After I received these error messages I placed the BJ at the edge of the board but not overlaying it. 

[image]

ForrestErickson, 2024-04-27:
The enclosure designer (Either you or someone else in a "company") must provide the requirement for how much the power jack should overhang the edge of the PCB.  

 This overhang must be equal to the enclosure thickness plus the uncertainty on the edge of the PCB (when it is routed or cut by JCLPCB) and possibly a small extra clearance.

Enclosure thickness = ______
PCB edge tolerance = ______
Extra clearance = _______

Add these up for the overhang.

nk25719, 2024-04-27:
Assume enclosure thickness of 2 mm, 
JLCPCB tolerance is +- 0.1 mm

ForrestErickson, 2024-04-30:
Regarding "Clearance: refers to the distance between the enclosure and the PCB."
Yes it does. I also called the same distance the overhang of the connector. 

## Example of Flush Mounted Connectors
[image]

ForrestErickson, 2024-05-14:
We do not have an enclosure which drives any other requirements and so this has been fixed (To the extent possible at the present. Had we an enclosure to which we had to design we would need to place so that the power barrel jack surface was correct for the enclosure.)

Closing.

closed 2024-05-14 by ForrestErickson