PubInv/PMD_-pseudo-medical-device-aka-Homework2- #5: No Copper in Antenna Radiation Pattern

ForrestErickson, 2024-04-23:
# Summary
There must be no conductor in the antenna radiation pattern.

The 20240421 has some copper which must be removed.
Here is the bottom layer:
[image]

nk25719, 2024-04-27:
I extended the keepout zone area around the antenna. 
(No more copper in the front and in the back layers)

[image]

ForrestErickson, 2024-05-14:
This has been fixed. 
[image]

Closing.

closed 2024-05-14 by ForrestErickson