PubInv/PMD_-pseudo-medical-device-aka-Homework2- #10: Set Current Limiting Resistors for LEDs

ForrestErickson, 2024-04-27:
Set the resistors for about 20 mA current.
This should be about 60 Ohms.

nk25719, 2024-04-28:
ESP32 GPIO pin outputs 3.3 volts.
LEDs are connected in such a way that the cathode (the shorter leg) is connected to ground.
LEDs have a forward voltage drop across them. It depends on the LED color and type. Red LED has a constant voltage drop of 2.04V.
The Red LED operated with a forward current of 20 mA to ensures optimal brightness and longevity.

The voltage across the resistor is the remaining voltage after subtracting the LED's forward voltage drop from the supply voltage: 3.3 - 2.04 = 1.3 volts.
The resistance value needed to limit the current flowing through the LED to the desired 20 mA given the voltage across the resistor:  R= 1.3/0.02 = 65 Ohms.

[image]

>> This screenshot depicts the connection of the red LEDs and the resistors to the GPIO pins of the ESP32, along with an accurate calculation of the resistor values connected to the GPIO pins and the red LEDs.

closed 2024-04-28 by nk25719

ForrestErickson, 2024-04-29:
## Lee's Bad.
Oh dear, I forgot to warn you to look for JLCPCB part numbers.
Where you will encounter the problem that not all resistor values are available, at least at low cost. 
There are several sets of "standard" values available about which to learn.
And the "standard" values will seem crazy.
There is also some history behind resistor values. 

## Real World Component Selection
When choosing all components, you want to choose parts that are common and therefore readily available and therefore inexpensive.
As you develop history you will want to start your choices, where possible, with parts you already have on hand. 

Read about the "E" series history and in particular try to use 1R, 10R, 100R, 1K, 10K, 100K and 1M where possible. For many digital systems you can do so.
However inevitably you will encounter more values in reference designs.
Try to use values from the 5% series, the E24 if possible. 
See: https://en.wikipedia.org/wiki/E_series_of_preferred_numbers#E24_subsets

## Inevitable To Learn About Standard Values
You will have an educational experience if you look up the price for similar parts but on the different "E" series.
I think that using JCLCPB will be difficult for this experience so use Digikey.com for example.
You will get a more realistic understanding of price, you should look up and compare the "Tape and Reel" price for each value. 

## Not Just Resistors
Staying with "standard" values is even more important with capacitors and very very more important with inductors.

Hope this helps.