PubInv/PMD_-pseudo-medical-device-aka-Homework2- #11: Set Switch Pull Up Resistors to 10K

ForrestErickson, 2024-04-27:
At all four locations.

nk25719, 2024-04-28:
To provide an input current of 0.3 mA to the GPIO input pins of the ESP32, a pull-up resistor of 10 kΩ is connected.
The resistor is connected in such a way that one side is connected to the 3.3 V input, and the other side is connected to both the switch and the GPIO pin of the ESP32.
When the switch is open, the resistor pulls the GPIO pin's voltage up to the 3.3 V supply, ensuring that the pin reads a logical HIGH or 1. When the switch is closed, it provides a path to ground, effectively pulling the voltage at the GPIO pin down to 0 V, resulting in a logical LOW or 0. So, the pull-up resistor ensures that the GPIO pin has a defined voltage level when the switch is open and allows the switch to control the state of the pin when it's closed.

[image]  

>> The screenshot shows the connection of the resistor, where one side is linked to the 3.3 V input, and the other side is connected to both the switch and the GPIO pin of the ESP32.

closed 2024-04-28 by nk25719

nk25719, 2024-04-28:
[image]
>> A screenshot of the updated scheatic as of 20240428