PubInv/PMD_-pseudo-medical-device-aka-Homework2- #12: Place Capacitors For Better High Frequency Effectives.

ForrestErickson, 2024-04-27:
## Three terminal regulator Circuit
Locate C4, C5, C6 and C7 near U1. The Small Value Capacitors Nearest U1
And move all of the parts near the power barrel jack.

Similar to this:
[image]

## ESP32 Power Supply Filtering
Move C1 and C3 near the power and ground pins of ESP32.  Ground pin 2 on ESP32.

Similar to this:
[image]

nk25719, 2024-04-28:
[image]  

> A screenshot of the relocation of the capacitors.

ForrestErickson, 2024-06-09:
Although this application notes use only one each capacitor on the input and output of the three terminal regulator, an example for component placement and trace routing can be found  on page 17 figure 27. 

https://www.ti.com/lit/ds/symlink/lm7800.pdf?ts=1717929189344&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FLM7800

Before making any changes to the component placement and routing, we should discuss what happens when two capacitors are placed in parallel for some insight.

nk25719, 2024-06-22:
[image]

> Capacitors are placeds for a better high frequency, and to minimize stray capacitance.

closed 2024-06-22 by nk25719

ForrestErickson, 2024-06-22:
Regarding, "and to minimize stray capacitance."
It is to minimize stray inductance.  Inductance is lead length.  A discussion of how to visualize magnetic field around a conductor is a rabbet hole down which we might go some time.

nk25719, 2024-06-23:
I noteed this down. thank you for the correction.