PubInv/PMD_-pseudo-medical-device-aka-Homework2- #14: Possible Grid Problem with ESP32 Footprint

ForrestErickson, 2024-04-27:
Pins may not be on 0.050 grid.
Place the pin number text avoiding lines.

Example where the flag could not be placed.
[image]

nk25719, 2024-04-28:
The ESP32 symbol was initially created on Grid 0.0050 in (0.127 mm).

As of 20240428, the ESP32 symbol has been edited and is now on Grid 0.0500 in (1.2700 mm).

[image]    

 >> In the screenshot of the updated ESP32 symbol, we can see the library name and file name displayed to the left of the ESP32, and the grid selection shown to the right of the ESP32.

closed 2024-04-28 by nk25719

ForrestErickson, 2024-04-29:
Good work. 
I feel vindicated warning of the nightmare of non standard grids.