PubInv/PMD_-pseudo-medical-device-aka-Homework2- #7: Warning upon Branch Renaming

nk25719, 2024-04-27:
We renamed the branch from Net-Classes to PCB-Barrel-Jack-ESP32-PIN-Names 

## This is the message we received from github upon renaming the branch
>> and we are capturing it in this issue incase we need it in the future

 
Renaming this branch will not update your members' local environments.

Your members will have to manually update their local environments. We'll let them know when they visit the repository, or you can share the following commands.

  git branch -m PCB-Barrel-Jack-ESP32-PIN-Names PCB-Barrel-Jack-ESP32-PIN-Names
  git fetch origin
  git branch -u origin/PCB-Barrel-Jack-ESP32-PIN-Names PCB-Barrel-Jack-ESP32-PIN-Names
  git remote set-head origin -a
  

[image]

ForrestErickson, 2024-05-14:
@nk25719 
Has this created any problems?
Is there something to investigate further?

If not this could be closed.

nk25719, 2024-08-11:
This has not created any problems and nothing to invastigate here. closing issue.

closed 2024-08-11 by nk25719