PubInv/PMD_-pseudo-medical-device-aka-Homework2- #15: The switches and the LEDs do not have a clear function as of 20240504

nk25719, 2024-05-04:
it is better if we add a silkscreen to specify the funcion of each of the switches and the LEDs

nk25719, 2024-05-04:
Switches were given sertain function.
Silkscreen is added to indicate these functions. 

[image]

nk25719, 2024-05-04:
Functions of the LEDs are added.
Silkscreen is added to indicate these functions.

[image]

ForrestErickson, 2024-05-14:
Hypthetical function names added to switches and LEDs on silk screen.

Closing

closed 2024-05-14 by ForrestErickson