PubInv/PMD_-pseudo-medical-device-aka-Homework2- #19: Uncertainty on Pre-Defined Via Sizes

ForrestErickson, 2024-05-21:
Lee left the following question on the FaceBook KiCad page.

> Regarding, via "pre-defined size"
> 
> Can someone suggest a family of good sizes?
> 
> I will be targeting for manufacture at JLCPCB.
> The minimum is trivial (The hole must not be smaller thant the smalles allowed. The annular ring must not be smaller than the smallest allowed). We have to respect the JLCPCB requirements. 
> 
> But what about progressively larger vias?
> 
> How do I make choices that do not have unexpected negative consequences?

ForrestErickson, 2024-05-29:
Question left with JLCPCB on line help on 20240529 

> Hello
> Regarding my setup for PCB design drills...
> I would think that sized that are some "standard" drill would be the best to choose but I have no informed idea what those sizes may be.
> Would the be number drill sizes?
> Increments of some metric dimension to which I living in the USA have never had exposure?
> Even some increment of mills?
> Read
> FYI I hope to share some list of prefered hole sizes (Drill sizes) with fellow designers of PCBs on the KiCAD Facebook group I frequent and where I asked this question.
>

ForrestErickson, 2024-05-29:
This linked in thread has some information on PCB hole sizes: https://www.linkedin.com/pulse/what-production-hole-diameter-pcb-rayming-techonloy-0vl5c/
There is a chart there:
https://media.licdn.com/dms/image/D5612AQHkzJ1s9hjzHQ/article-inline_image-shrink_1000_1488/0/1699238149589?e=1722470400&v=beta&t=GxRSd8Mz3mJPI1Iur5hHvuaNLWC-nTDxOFG2-RS0GaU

Another reference but with far too many to help:  
https://www.kyoceraprecisiontools.com/pcb/pdf/PCB_Diameter_Chart.pdf

nk25719, 2024-08-11:
This is a screenshot for the micro vias and holes meseasurements we used in this projects
> 
[image]

closed 2024-08-11 by nk25719