PubInv/PMD_-pseudo-medical-device-aka-Homework2- #20: Add teleportation feature, random location

nk25719, 2024-05-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
On days when I get bored I would ike to do some sight seeing.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
When I hold the "S1-morseCodeDanger1" and the "S2-RepeatCalculation1" simultaneously for 10 seconds, I transport to a random location.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
When I hold the "S1-morseCodeDanger1" and the "S2-RepeatCalculation1" simultaneously for 10 seconds, the device will materialize a playfull kitten.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]
[image]

ForrestErickson, 2024-05-25:
What could possibly go wrong?

nk25719, 2024-06-01:
Photo shows that we can materialize a kitten, can we do a puppy?

nk25719, 2024-10-12:
This issue was to test the issues and the procedure of adding features for our project, using random jokes we trained how to do that. 
cc: @ForrestErickson 

Closing..

closed 2024-10-12 by nk25719