PubInv/PMD_-pseudo-medical-device-aka-Homework2- #27: PLACE JLCPCB ORDER

nk25719, 2024-07-05:
Update the files of the finalized PCBA for the 'esp32-6leds-2switches-1pj-circuit' and place an order on JLCPCB.

nk25719, 2024-07-05:
Add an extra connector to the board, add an extra power jack to the board, and update the footprint and the schematic before placing the order.

nk25719, 2024-07-09:
240709 Cleared al the schematic, 
Error and warning too 

[image]

ForrestErickson, 2024-07-09:
Place signal names on I2C connector and UART connector.

nk25719, 2024-07-09:
[image]

> PCB Editor DRC Clreared

ForrestErickson, 2024-07-09:
##  "mixed values" is indicative of an error.
The Bulk Editor shows "mixed values" on several locations which should be fixed.

[image]

ForrestErickson, 2024-07-09:
##   Exclude Test Points
For the Test Points we do not have a JLCPCB number. Was that on purpose?

1. If it is on purpose then exclude them from the BOM.  
2. If not on purpose then find the correct JLCPCB part number and update the schematic.

[image]

ForrestErickson, 2024-07-09:
## Exclude Mounting Holes
They are a board feature, not a part so exclude from BOM / Parts List

[image]

ForrestErickson, 2024-07-09:
## Capacitors and Others Missing Description
Fore example C2
See below for more:
[image]

nk25719, 2024-07-10:
<html>
<body>

R8 | 62 | Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder | RTT031000FTPC103197
-- | -- | -- | --

</body>
</html>

nk25719, 2024-07-10:
[image]

ForrestErickson, 2024-07-10:
Please commit all of the files from work on 20240710.

ForrestErickson, 2024-07-12:
## Summary For PCBWay Compatibility

To the existing fields of part numbers a new field for PCBWay Part numbers must be added.   
Do not overwrite the JLCPCB part numbers.

nk25719, 2024-07-13:
> Please commit all of the files from work on 20240710.

The files are added.

ForrestErickson, 2024-07-16:
@nk25719 
Please add into this issue the message we got from JLCPCB regarding can not ship to Lebanon.
Please add into this issue the notes / correspondence with PCBWay regarding the BOM and other file compatibility.

ForrestErickson, 2024-07-17:
## JLCPCB Order Journal

I have found BOM files at: \KiCad-esp32-6leds-2switches-1pj-circuit\JLCPCB20240710

Lee started setting up a JLCPCB order for USA.
The 62 ohm resistor in our BOM was insufficent stock so I substituted for this PN: C308152

Component placement preview:
[image]

Order ready:
[image]

As ordered:
[image]

ForrestErickson, 2024-07-19:
---------- Original Message ----------
From: JLCPCB <noreply@jlcpcb.com>
To: forresterickson@comcast.net
Date: 07/17/2024 8:17 PM EDT
Subject: Please confirm production file of your JLCPCB order

Dear ForrestErickson,
The production file of your order completed. Your order 2999181A_Y16 is ready for manufacturing now.

To avoid production delays, please check and confirm the production file through your [order history](https://mandrillapp.com/track/click/31015052/jlcpcb.com?p=eyJzIjoiU0dYUXhXa2JoSjZRa0RURml0WmN6cmJieHVBIiwidiI6MSwicCI6IntcInVcIjozMTAxNTA1MixcInZcIjoxLFwidXJsXCI6XCJodHRwczpcXFwvXFxcL2psY3BjYi5jb21cXFwvdXNlci1jZW50ZXJcXFwvb3JkZXJzXFxcL1wiLFwiaWRcIjpcIjVjOWNjN2E2M2ViODRhNmZiMDFmOGYzM2JiOGI3YzI4XCIsXCJ1cmxfaWRzXCI6W1wiNGJhOThhMTA2YzBmYTI0NzcwZjIxNGNmNzQ0NmNhMzk0NDFiMjJjN1wiXX0ifQ) ASAP.

This is an automatically generated email – please do not reply to it. If you have any queries regarding your order please email [support@jlcpcb.com](mailto:support@jlcpcb.com). 

Sincerely,

JLCPCB Support Team

I am confirming the order. First of the PCBs.
However the PWA (printed wiring assembly) analysis shows a problem.

Zoom in on the PCB

From this preview, It looks like C7 and C9 need a larger body part.

However when I look up the part, C2895272, it is a 5x5.4 
Asking the on line support for help.

This part might also work:
VS1C470MC054000CE0
Manufacturer  Changzhou Huawei Elec
MFR.Part #  VS1C470MC054000CE0
JLCPCB Part #  C240546
Package  SMD,D5xL5.4mm
Description  47uF 16V ±20% SMD,D5xL5.4mm Aluminum Electrolytic Capacitors - SMD ROHS

We resolved the placement of the three capacitors and the polarity of D2.
I think the order is proceeding.

Lee

nk25719, 2024-07-20:
> @nk25719 Please add into this issue the message we got from JLCPCB regarding can not ship to Lebanon. Please add into this issue the notes / correspondence with PCBWay regarding the BOM and other file compatibility.

[image]

Shipping from JLCPCB to Beirut, Lebanon is not available. 

**PCBWAY is another manufacturer and they ship to Lebanon, Issue #30 for ref.**

ForrestErickson, 2024-08-31:
Order received in Maryville.
[image]

Closing.

closed 2024-08-31 by ForrestErickson