PubInv/PMD_-pseudo-medical-device-aka-Homework2- #31: Using Multitasking, Make so that when one button is Pressed and Held One LED changes the rate at which it blinks.

ForrestErickson, 2024-07-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Make all LEDs blink at independent rates and duty cycles
Press and HOLD a button and one and only one LED blinks faster

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ForrestErickson, 2024-07-20:
Step two, Press and release the button and the frequency will change.
Press and release a second time and the frequency changes back.

nk25719, 2024-08-11:
FIRMWARE/HOLDDownS1toBlinkLEDpin2Faster_jul27a/HOLDDownS1toBlinkLEDpin2Faster_jul27a.ino
**when we hold down S1 LED on pin2 blinks faster**

nk25719, 2024-08-12:
One button is Pressed and Held One LED changes the rate at which it blinks :

```
#include <Arduino.h>

const int ledPin = 2;          // LED pin
const int buttonPin = 36;      // Button pin

unsigned long previousMillis = 0;  // To store last time LED was updated

// Timing for the LED blink rates
const unsigned long normalOnTime = 500;
const unsigned long normalOffTime = 500;
const unsigned long fastOnTime = 100;
const unsigned long fastOffTime = 100;

unsigned long onTime = normalOnTime;  // Current on time
unsigned long offTime = normalOffTime;  // Current off time

bool ledState = LOW;  // Initial LED state
bool buttonState;     // Current button state
bool lastButtonState = HIGH;  // Last button state (assuming pull-up)

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(buttonPin, INPUT_PULLUP); // Assuming the button is connected to GND
  Serial.begin(115200);
}

void loop() {
  unsigned long currentMillis = millis();
  buttonState = digitalRead(buttonPin);

  // Check if the button state has changed from HIGH to LOW (button press)
  if (buttonState == LOW && lastButtonState == HIGH) {
    // Toggle fastBlink mode
    if (onTime == normalOnTime) {  // If currently in normal mode, switch to fast
      onTime = fastOnTime;
      offTime = fastOffTime;
    } else {  // If currently in fast mode, switch to normal
      onTime = normalOnTime;
      offTime = normalOffTime;
    }
    delay(50); // Debounce delay
  }
  lastButtonState = buttonState;  // Update the last button state

  // Check to update LED state based on the current time and on/off durations
  if ((ledState == HIGH && (currentMillis - previousMillis >= onTime)) || 
      (ledState == LOW && (currentMillis - previousMillis >= offTime))) {
    ledState = !ledState;  // Toggle the state
    digitalWrite(ledPin, ledState);  // Update the actual LED
    previousMillis = currentMillis;  // Reset the timer
    Serial.println(ledState ? "LED turned on" : "LED turned off");
  }
}

```

ForrestErickson, 2024-08-13:
# Testing Report
## Test Setup Description
Lee used WOKWI to test the code above by repurposing an existing WOKWI schematic and then modifying the sketch above for a different input pin for the button (GPIO0).
There are several LEDs in the schematic.

NOTE: Code review reveals that only one LED is controlled by this sketch 

The code compiled and ran.
The LED blinks slowly.
Press the button and the LED blinks faster.
In the simulation the 

- slow blink was about 3 seconds though from the code it should be about 1 second.
- the fast blink when button held was faster at less than a second.

Lee modified the code to print a time stamp of the simulated mills and confirmed the blink rates are as intended.
Lee Also modified the code so that the fast blink was asymmetric on / off times.

## Simulations Screen shots

Slow Blink  
[image]

Fast Blink  
[image]

## Test Results
**The code satisfies the requirements stated above for changing the rate of blink of the LED by pressing and holding the button.**

## Other, Unnecessary Delay
The code " delay(50); // Debounce delay" is unnecessary and since it uses a delay() function violates the idea of real time multitasking.

nk25719, 2024-08-13:
Regarding:

> ## Other, Unnecessary Delay
>The code "delay(50); // Debounce delay" is considered unnecessary because it uses a delay() function, which violates the principle of real-time multitasking.

This was initially intended to prevent accidental re-triggering of the switch. A debouncing delay in the prototype help us avoid unintended triggers. However, debouncing has also been handled using the millis() method, and the code has been updated accordingly:

Nagham would like to have the code that Lee modified so she can adjust the debouncing delay in it.

nk25719, 2024-08-13:
```
#include <Arduino.h>

const int ledPin = 2;          // LED pin
const int buttonPin = 36;      // Button pin

unsigned long previousMillis = 0;  // To store last time LED was updated
unsigned long lastDebounceTime = 0;  // the last time the output pin was toggled
const unsigned long debounceDelay = 50;  // the debounce time; increase if the output flickers

// Timing for the LED blink rates
const unsigned long normalOnTime = 500;
const unsigned long normalOffTime = 500;
const unsigned long fastOnTime = 100;
const unsigned long fastOffTime = 100;

unsigned long onTime = normalOnTime;  // Current on time
unsigned long offTime = normalOffTime;  // Current off time

bool ledState = LOW;  // Initial LED state
bool buttonState;     // Current button state
bool lastButtonState = HIGH;  // Last button state (assuming pull-up)

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(buttonPin, INPUT_PULLUP); // Assuming the button is connected to GND
  Serial.begin(115200);
}

void loop() {
  unsigned long currentMillis = millis();
  bool readButton = digitalRead(buttonPin);

  // Check for button press (assuming low on press)
  if (readButton != lastButtonState) {
    lastDebounceTime = currentMillis;
  }

  if ((currentMillis - lastDebounceTime) > debounceDelay) {
    // if the button state has changed:
    if (readButton != buttonState) {
      buttonState = readButton;
      // only toggle the LED if the new button state is LOW
      if (buttonState == LOW) {
        if (onTime == normalOnTime) {  // If currently in normal mode, switch to fast
          onTime = fastOnTime;
          offTime = fastOffTime;
        } else {  // If currently in fast mode, switch to normal
          onTime = normalOnTime;
          offTime = normalOffTime;
        }
      }
    }
  }

  lastButtonState = readButton;  // Update the last button state

  // Check to update LED state based on the current time and on/off durations
  if ((ledState == HIGH && (currentMillis - previousMillis >= onTime)) ||
      (ledState == LOW && (currentMillis - previousMillis >= offTime))) {
    ledState = !ledState;  // Toggle the state
    digitalWrite(ledPin, ledState);  // Update the actual LED
    previousMillis = currentMillis;  // Reset the timer
    Serial.println(ledState ? "LED turned on" : "LED turned off");
  }
}

```

ForrestErickson, 2024-08-13:
Regarding, "Nagham would like to have the code that Lee modified so she can adjust the debouncing delay in it.  "
Find code at: https://wokwi.com/projects/406072197680452609

Notice I have no delay for debounce.
I also changed the pin for input.  Why did I choose the pin which I chose?

ForrestErickson, 2024-08-14:
I guess I should note that the task as written for holding a button was to avoid the requirement for debouncing at this time to consnetrate on the Multitasking aspect of programing.  

If you want to investigate debouncing I suggest that an independant sketch be developed which first of all is prone to or vulnerable to poor switch debouncing. Only then is it time to develop a debounce counter measure.

In the hardware world this would mean getting a real switch with real bouncing behavior. 
How about in the simulated world? 

In the short term work on multi tasking many LEDs (and then servo motors).

ForrestErickson, 2024-09-02:
@nk25719 

Perhaps Tuesday 3 September, we should review the status of this issue and the firmware you have written.

nk25719, 2024-10-27:
HOLDDownS1toBlinkLEDpin2Faster_jul27a.ino

when S1 is clicked once the blinking of the LED on pin 2 increase in rate until S1 is pressed again. 

cc: @ForrestErickson

closed 2024-11-22 by nk25719