PubInv/PMD_-pseudo-medical-device-aka-Homework2- #33: Assemble PCBWAY Order of 202407

ForrestErickson, 2024-08-31:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
Capture the out of box experience of receiving the PCBWAY Order of 202407 including finish assembly

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Capture of every step.

nk25719, 2024-09-01:
### Journal of Assembly 

The shipping package
[image]

Inside the Shipping Package
[image]

Wrapping of the PCBAs
[image]

Unwrapping of Assemblies
[image]

Unpacking Assemblies
[image]

Closer Look, Top View 
[image]

Closer Look, Bottom View 
[image]

Checking Assemply, Wrong pin spacing on DIP for ESP32
[image]

## Containment of Pin Spacing 

[image]

connected the ESP32 and Assembled the LEDs

nk25719, 2024-10-12:
**Factory Test with Firmware**

To verify the hardware assembly, several factory test codes were used:

- **Marching One and Marching Zero test codes** were used to test the LEDs.
- Switch functionality was successfully tested.
- The power jack was tested and confirmed to be working.
- The ESP32 was mounted on the board (the dimensions were not compatible, so the ESP32 legs were modified to fit).
- The connectors for the LCD and DFPlayer were tested, with positive results.

**Conclusion**: The assembly is properly designed, and the PCB functions as expected.

[image]

Closing..

closed 2024-10-12 by nk25719