PubInv/PMD_-pseudo-medical-device-aka-Homework2- #34: Retain the PCBA assemblies of 202407 in Rev1 enclosure and check the how

nk25719, 2024-09-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
_We do not know what screw is going to mount the PCBA into the enclosure_ 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
_We need the exact manufacturer's name and part number and the distrebuter name and the distrebuter part number.
the diameter holes of the four corners are 3.5 mm_ 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
_Design our own screws. or Use glue or Use tape_

**Additional context**
Add any other context or screenshots about the feature request here.

samoyin, 2024-09-07:
I found a screw with a 3.5mm diameter hole.
Here are the Specs.
It's a M3.5-0.60 Phillips Pan Head Machine Specs: DIN 7985A / ISO 7045. The distributors are Frontec screws.
Other information can be found here. [https://monsterbolts.com/products/mach-phil-pan-blk-m3-5?variant=21508467523667]

ForrestErickson, 2024-09-07:
@samoyin 

Regarding, "I found a screw with a 3.5mm diameter hole."
A 3.5mm screw into a 3.5mm hole is not guaranteed to have clearance. May have negative or interference clearance. 

For example. The PCB material is drilled to 3.5 mm nominally. It may be under sized. Then the PCB has copper plated on to the hole. It is now defiantly smaller still. 
Even if the screw exactly fits through the hole, on one of the four holes there are four total and the thermal expansion and humidity expansion of the PCB and the enclosure may not be the same and so all four holes and screws may be able to fit at one combination of temperature and humidity but not others.

It is also difficult for an operator to place screws through a hole with zero clearance.

There is a vocabular about fit. See this document for some insight on clearance for holes:
https://fpg.phys.virginia.edu/fpgweb/useful_info/Clearance_Hole_Drill_Sizes.pdf

With experience you want to learn how to never specify unnecessarily tight fit or unnecessary precision.

nk25719, 2024-10-27:
[image]

-----

3.2mm diameter and 1.9mm height screw
with cap of 6.4mm diameter and .99mm height
[image]

cc: @ForrestErickson 

Closing.

closed 2024-10-27 by nk25719