PubInv/PMD_-pseudo-medical-device-aka-Homework2- #36: Enhance the Factory Test Firmware

nk25719, 2024-09-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
_The current Test fails to detect shorts  between LEDs_
_Opperate every LED independantly of every other LED_

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
_Adding a Marching 1 and a marching 0_ 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ForrestErickson, 2024-09-02:
I decided to ask ChatGPT for some coding ideas.  It returned this which I have not tried:

https://chatgpt.com/share/8afc3085-eb42-4212-ace3-92c104a99e49

nk25719, 2024-09-03:
https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/tree/main/FIRMWARE/marchingOneTest

https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/tree/main/FIRMWARE/marchingZeroTest

cc: @ForrestErickson

closed 2024-09-03 by nk25719