PubInv/PMD_-pseudo-medical-device-aka-Homework2- #38: Hardware Bug,, PCB design error 20240421, J2 J3 spacing incorrect

nk25719, 2024-09-01:
**Describe the Failure / Bug**
A clear and concise description of what the failure / bug is.
Short Summary of the expected behavior:
Short Summary of the buggy behavior:
_The Spacing between J2 and J3 is not compatable with the width of the ESP32 devkitV1 we are using,

**Device Under Test, aka DUT**
Steps to reproduce the behavior:  
1. Model Number:   20240421
2. Serial Number(s):  { USA1 USA2 USA3 USA4 USA5 LEB1 LEB2 LEB3 LEB4 LEB5 }
3. Location where failure / bug occurred: Maryville and Aley
4. Hardware configuration: not applicable
5. Extent of Failure: 
** Is failure intermittent? no
** Is failure fixed by power cycling? no
** Is failure perminant? yes

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

nk25719, 2024-10-12:
## Adjusted ESP32 Pins

The pins of the ESP32 were bent inward to fit into connectors J2 and J3.

[image]

Lesson Learned:
The width of ESP32 modules can vary. We need to adjust the spacing measurements of J2 and J3 to accommodate these differences. 

Closing..

closed 2024-10-12 by nk25719

ForrestErickson, 2024-10-13:
@nk25719 
We should have followed the project outline you can find at this "Best Widget Ever" repository I made, the bwe.
https://github.com/ForrestErickson/bwe

What new detail could be added to this process to better help us catch this error?