PubInv/PMD_-pseudo-medical-device-aka-Homework2- #39: 3D Models Incomplete, Path Problem on Some

ForrestErickson, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The 3D view of the printed wiring assembly is incomplete
[image]
For example the power jack is not represented.
Also missing is the very important ESP32 Kit

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
3D models for all parts on PCB and the ESP32 Kit so that a full 3D STEP file can be exported to aid enclosure design task.

**Additional context**
_Add any other context or screenshots about the feature request here._

Note for the Power Jack there is an error in the path to the 3D model the root cause of which should be investigated
[image]

ForrestErickson, 2024-09-02:
Note: Generation of STEP file reports the many missing 3D models

[image]

Here is the resulting 3D file (which I have not and will not check in) opened in FreeCAD A2plus work bench.
[image]

CC: @samoyin

nk25719, 2024-10-27:
Issee #67 shows how we fixed the Path Problem for the 3D models of the parts. 

[image]

cc: @ForrestErickson 

Closing.

closed 2024-10-27 by nk25719