PubInv/PMD_-pseudo-medical-device-aka-Homework2- #40: From KiCad, Export a STEP (? other types) file of Printed Wiring Assembly, May Help Enclosure Design

ForrestErickson, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

Problem, For design of an enclosure, a CAD file of the Printed Wiring Assembly will help.
With a 3D file fit into the enclosure may be tested, and confirmed.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
The PCB designer should export from KiCad the assembly in a STEP file format (Or any other useful format)

**Additional context**
_Add any other context or screenshots about the feature request here._

Here is screen shot of the path to file export
[image]

Here is the STEP export dialog
[image]
There are several 3D model types. Choose which ever works best for the user of this file (the person designing the enclosure for example, Trial and experimentation may be required.)

Note the options for setting the Coordinates. Choose which ever works best for the user of this file (the person designing the enclosure for example, Trial and experimentation may be required.)

ForrestErickson, 2024-09-02:
@samoyin 
Are you experienced enough with OpenSCAD to know if it is possible to import a STEP file to help your work?

I use FreeCAD.
In FreeCAD I can import into an assembly work bench the STEP file generated by KiCad. Please see the issue: https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/41

CC: @nk25719

samoyin, 2024-09-07:
Hello Lee,
 I have not done this before but there is a way to import a STEP file to OpenScad and convert the Step file to an OpenScad program. Also, I saw your most recent email. I hope you rest and recover well from the surgery. See you when you get better!!!
From Oyin.

nk25719, 2024-10-27:
From Kicad project manager we navagate to the PCB editor and we click on file then export and we chose the STEP file format to do an export, we chose the path we want to export the STEP file into and we do the export:

1.  [image]

2.   [image]

-----

### **The STEP file format of the Homework2 PCBa**
[image]

closed 2024-10-27 by nk25719