PubInv/PMD_-pseudo-medical-device-aka-Homework2- #41: Capture 3D Models for The Assembly of This Product, ESP32 and More

ForrestErickson, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
Gather useful 3D models into the project in an organized way for this project and for the Krake (and other Public Invention projects)

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
**FOR EXAMPLE**, Here is a STEP file of the Ai-Thinker Nodemcu-32s (ESP32 Development Board)
https://grabcad.com/library/ai-thinker-nodemcu-32s-esp32-development-board-1
CAUTION: Unlike the ESP kit for our PCB, This model has 19 pins on a side. and will not work as is.
This should be captured into the github project in an organized way. Clutter may develop if the result of our development of a lot of 3D models.  We should anticipate where to put this clutter keeping in mind future versions may evolve. 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
NONE
**Additional context**
_Add any other context or screenshots about the feature request here._

Here I have imported the  Ai-Thinker Nodemcu-32s (ESP32 Development Board)  and placed it into the sockets on the assembly.  
[image]
**Notice:** For this Ai-Thinker Nodemcu-32s (ESP32 Development Board) model, you can see that the spacing between pin rows is correct but the number of pins is incorrect

ForrestErickson, 2024-09-02:
Here is a 3D model of a Kit 1 which I think is physicaly the same as the kits Nagham and I have.
https://grabcad.com/library/esp32-dev-kit-v1-1

Here I have placed the STP model into the assembly

[image]
We can see the pin number is correct

And we can see the pin spacing is incorrect
[image]

ForrestErickson, 2024-09-02:
To down load files from https://grabcad.com you (Nagham, Oyin) will need to make an account for yourselves if you have not already.

nk25719, 2024-10-27:
Oyin found a 3D model for esp32 with 15 pins 

We need to discuss if it is okay to scale the model, 
Nagham nows that the footprint should be 1 to 1 scale with the real model 

Here is the 3D model that I added to esp32 footprint 
( scaled the y to .99)

[image]

> 
> 
> download the step file of the esp32 15 pin through: (https://forum.kicad.info/uploads/short-url/ozpprxF4vqvKlNpxuIEfuTLoVdP.zip)
> 
> link to the article: https://forum.kicad.info/t/schematic-esp-32s-30-pin/48168/3

cc: @ForrestErickson

nk25719, 2024-10-28:
The 3D model is correct; however, the footprint width measurement is designed for an 18-pin ESP32 module rather than a 15-pin one. 
Adjusting the footprint isn’t feasible at the moment, so a RevA will be necessary. Scaling the Y-axis was used as a workaround to represent placing the ESP32 on the Homework2 board.

nk25719, 2024-11-02:
With the wong spacing of J1 and J2 scaling the esp32 is the best way to depict the situation currenty after we successfully fixed it with resizing the pins of the esp32 module legs. 

cc: @ForrestErickson 

Closing.

closed 2024-11-02 by nk25719