PubInv/PMD_-pseudo-medical-device-aka-Homework2- #43: Incomplete STEP File Export

nk25719, 2024-09-07:
Loading board
Determining PCB data
Board Thickness from stackup: 1.580 mm
Build STEP data
Add component R14.
Could not add 3D model to D6-lostSock1.
File not found: 
Could not add 3D model to S2-RepeatCalculation1.
File not found: 
Add component J2.
Add component J5_DFPLAYER1.
Add component D2.
Add component TP5.
Add component R12.
Add component TP2.
Add component R2.
Add component C2.
Add component C4.
Add component C3.
Add component F1.
Add component R9.
Add component R10.
Add component R4.
Add component C1.
Could not add 3D model to D1.
File not found: 
Add component R3.
Could not add 3D model to D3-babyCrying1.
File not found: 
Could not add 3D model to S1-morseCodeDanger1.
File not found: 
Add component R1.
Add component C7.
Add component TP4.
Add component R7.
Could not add 3D model to D5-coldFood1.
File not found: 
Add component TP3.
Add component R5.
Could not add 3D model to D9-MUTE1.
File not found: 
Could not add 3D model to S3-muteFiveMin1.
File not found: 
Could not add 3D model to D4-ONOFF1.
File not found: 
Add component J4_LCD1.
Could not add 3D model to J1.
File not found: 
Add component C6.
Add component R6.
Add component C5.
Could not add 3D model to D7-SheildFailure1.
File not found: 
Add component R11.
Add component U1.
Add component R8.
Add component C8.
Add component R13.
Could not add 3D model to S4-SendEmergMessage1.
File not found: 
Add component C9.
Add component J3.
Could not add 3D model to D8-highBP1.
File not found: 
Add component TP1.
Create PCB solid model
Board outline: find 4 initial points
Build board main outline 1 with 4 points.
Build board cutouts and holes (116 hole(s)).

Generate board full shape.
Writing STEP file

STEP file 'C:\Users\Laptop Pro\Downloads\homeWork2.step' created.

nk25719, 2024-10-12:
cc: @ForrestErickson 

Please help me understand the purpose of this issue and its current status.

ForrestErickson, 2024-10-13:
@nk25719 
Use FreeCAD.
In for example a new an empty file, import the STEP file.
You will find that many of the parts are not represented with 3D models.
If you look at the KiCad 3D render they will be missing there too.

It is only a requirement to have 3D models for each part on the PCB if we choose it to be so.
We might for example want to put emphasis only on big, tall, parts which affect the enclosure design.
We rather certainly want all parts for which there are holes in the enclosure like LEDs, switches and jacks.

Should the design review process at Best Widget Ever be modified to reflect this thinking and if so how?

nk25719, 2024-10-27:
[image]

Screanshot of the STEP file with all the parts 3D models.

In issue #67 the missing 3D models issue was adressed.

closed 2024-10-27 by nk25719