PubInv/PMD_-pseudo-medical-device-aka-Homework2- #45: Inconsistant Naming, Wrong Naming of Button Pins Firmware versus Schematic Version 1 PCB

ForrestErickson, 2024-09-08:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
The name of GPIO pins on the schematic and PCB should match the naming in firmware.
Names should reflect function.

However for example S1 is called "S1-morseCodeDanger1" on the schematic and PCB and connected to GPIO36
[image]
[image]

But in the firmware is const int MUTE_BUTTON_PIN = 36; // GPIO 36 for the mute button

These names are inconsistent.
In all firmware other pins and pin names should be reviewed and corrected where errors are found.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

Change in all firmware to reflect the actual name on the schematic and PCB (and / or possibly Krake)

ForrestErickson, 2024-09-08:
Note the exact text on the schematic for the switches has a "-" which is not a character that can be used in a variable name.
Example: S1-morseCodeDanger1
So change to S1_morseCodeDanger1

### Lesson Learned
In future use only upper case letters and numbers and underscore for net names which will be referenced in firmware.

nk25719, 2024-09-09:
During the process of transferring the PMD into a Krake, the switches and the LED were renamed.

PMD inherrited MUTE_BUTTON_PIN = 36; // GPIO 36 for the mute button from Krake

nk25719, 2024-11-02:
cc: @ForrestErickson 

Closing.

closed 2024-11-02 by nk25719