PubInv/PMD_-pseudo-medical-device-aka-Homework2- #47: Notes on Assembly, LED Pattern During Reset

ForrestErickson, 2024-09-11:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

This issue is simply to capture the typical behavior observed on a KiCad-esp32-6leds-2switches-1pj-circuit assembly during the ESP32 reset condition.

## Typical Reset LED Behavior
[image]
Only the power LED near J1 and the D8 LED are illuminated. D8 is not full brightness

## Conclusion
If an assembly does not have this behavior suspect a hardware problem.

nk25719, 2024-09-12:
[image]

> LED 8 is connected to D5

[image]

> Captured pin D5 functionalities

ForrestErickson, 2024-09-12:
Lee found a schematic for by search on "schematic diagram for esp32 dev kit1 V1"
See: 
https://embedded-systems-design.github.io/overview-of-the-esp32-devkit-doit-v1/
Look for schematic at this URL:
https://embedded-systems-design.github.io/overview-of-the-esp32-devkit-doit-v1/SchematicsforESP32.pdf
Detail showing the 10K pull up on D5
[image]

### Conclusions
The soft illumination of the Diode D8 on the PMD when the system is in reset is behavior we can expect.
There is no dangerous (no latent) failures of any hardware likely.

Closing this issues as explained.
This information might be incorporated into a Factory Test document if one is ever written.

closed 2024-09-12 by ForrestErickson