PubInv/PMD_-pseudo-medical-device-aka-Homework2- #49: Hardware Bug - PCB  REV 20240421, J5 HAS 3V3 PWR, However DF Player Requirs 5V

nk25719, 2024-09-12:
**Describe the Failure / Bug**
A clear and concise description of what the failure / bug is.
Short Summary of the expected behavior:
Short Summary of the buggy behavior:
_connected the DF Player at J5. loaded the PMDtoKrake firmware that was known to work in the Mocking Krake, during running the code, the serial monitor_ 

[image]

**Device Under Test, aka DUT**
Steps to reproduce the behavior:  
1. Model Number:   
2. Serial Number(s):   LEB 1
3. Location where failure / bug occurred lebanon and texas 
4. Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.)
5. Extent of Failure
** Is failure intermittent? no
** Is failure fixed by power cycling? yes
** Is failure perminant? yes

**To Reproduce**
Steps to reproduce the behavior:
1. Go to:  https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/blob/main/FIRMWARE/PMDtoKrake/PMDtoKRake_KrakeV2/PMDtoKRake_KrakeV2.ino
2. Click on : Copy and upload the IDE code to the ESP32 module
3. Navigate to: Serial monitor pf the IDE
4. See error: DF player is not initializing 

**Expected behavior**
A clear and concise description of what you expected to happen.
DF Player initialized

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware description
Model:
Serial Number:
Other:

**Test Equipment
Model:
Serial Number:

**Additional context**
Add any other context about the problem here.

nk25719, 2024-09-12:
cc: @ForrestErickson

ForrestErickson, 2024-09-12:
## Journal Trying To Duplicate in Maryville TN USA

## Test hardware and Setup
Loading into device SN USA1 which does NOT have an LCD connected and does NOT have a DFPlayer connected.

1. Sync my fork
2. Pull to update local clone
3. Open PMDtoKRake_KrakeV2.ino in Arduino IDE, compile and upload into target hardware.
4. Run and capture the serial port traffic.  
[image]
5. With Android smart phone find the Soft AP "ESP32" and connect with success.
6. With Android smart phone web browser try to open 192.168.4.1 with OUT success. The browser reports, "This site can't be reached. 192.168.4.1 took too long to respond. Try: Checking the connection."
7. 

###  Failure to Duplicate DF Player Error
In Maryville, I cannot duplicated the error. I never see the text "Initializing DFPlayer"
Is the missing LCD or missing DFPlayer hardware the reason the message fails to duplicate?

@nk25719 
Please duplicate my setup by running your PMD with out the LCD and with out the DFPlayer connected. Do you get the same result as I outlined above or not?

ForrestErickson, 2024-09-12:
## More Notes On DFPlayer Testing

From https://wiki.dfrobot.com/DFPlayer_Mini_SKU_DFR0299 Lee found some code for evaluating the module
Here it is with modifications for the ESP32 pins as per the PMD.

```
//From: https://wiki.dfrobot.com/DFPlayer_Mini_SKU_DFR0299
//File saved as: DFPlayer_AllFunctions
/***************************************************
 DFPlayer - A Mini MP3 Player For Arduino
 <https://www.dfrobot.com/product-1121.html>

 ***************************************************
 This example shows the all the function of library for DFPlayer.

 Created 2016-12-07
 By [Angelo qiao](Angelo.qiao@dfrobot.com)

 GNU Lesser General Public License.
 See <http://www.gnu.org/licenses/> for details.
 All above must be included in any redistribution
 ****************************************************/

/***********Notice and Trouble shooting***************
 1.Connection and Diagram can be found here
<https://www.dfrobot.com/wiki/index.php/DFPlayer_Mini_SKU:DFR0299#Connection_Diagram>
 2.This code is tested on Arduino Uno, Leonardo, Mega boards.
 ****************************************************/

#include "Arduino.h"
#include "DFRobotDFPlayerMini.h"

#if (defined(ARDUINO_AVR_UNO) || defined(ESP8266))   // Using a soft serial port
#include <SoftwareSerial.h>
SoftwareSerial softSerial(/*rx =*/10, /*tx =*/11);
#define FPSerial softSerial
#else
#define FPSerial Serial1
#endif

DFRobotDFPlayerMini myDFPlayer;
void printDetail(uint8_t type, int value);

void setup()
{
#if (defined ESP32)
  //FPSerial.begin(9600, SERIAL_8N1, /*rx =*/D3, /*tx =*/D2);
  FPSerial.begin(9600, SERIAL_8N1, 16, 17); // rx2,  tx2 on ESP32 DevKit;
#else
  FPSerial.begin(9600);
#endif

  Serial.begin(115200);

  Serial.println();
  Serial.println(F("DFRobot DFPlayer Mini Demo"));
  Serial.println(F("Initializing DFPlayer ... (May take 3~5 seconds)"));

  if (!myDFPlayer.begin(FPSerial, /*isACK = */true, /*doReset = */true)) {  //Use serial to communicate with mp3.
    Serial.println(F("Unable to begin:"));
    Serial.println(F("1.Please recheck the connection!"));
    Serial.println(F("2.Please insert the SD card!"));
    while(true);
  }
  Serial.println(F("DFPlayer Mini online."));

  myDFPlayer.setTimeOut(500); //Set serial communictaion time out 500ms

  //----Set volume----
  myDFPlayer.volume(10);  //Set volume value (0~30).
[... 134 lines cut ...]
          Serial.println(F("In Advertise"));
          break;
        default:
          break;
      }
      break;
    default:
      break;
  }

}

```

When running on the PMD with out the DFPLayer connected the result is:
[image]

### Behavior of PMDtoKRake_KrakeV2.ino versus DFPlayer_AllFunctions
The DFPlayer_AllFunctions firmware reports the DFPlayer unable to begin. The PMDtoKRake_KrakeV2.ino firmware never gets as far as the DFPLayer even starting.  

## Conclusion. 
In Maryville, the PMDtoKRake_KrakeV2.ino is blocking in the delivery of the WiFi Manager.
The code in Aley may not be the same.

CC: @nk25719

ForrestErickson, 2024-09-12:
I have rewritten the **PMDtoKRake_Krake** setup() so that the DFPLayer initialization comes before the WiFi connection and now I get the same error of "DFPlayer Mini not detected:"
This code also reports the firmware name and version and some additional information to help us make sure we know what code we are testing.
I will commit the modified **PMDtoKRake_Krake** firmware

Screen shot of serial capture
[image]

ForrestErickson, 2024-09-21:
@nk25719 
As a result of our work today and conversation today I am not sure if you still think 3.3V to the DFPlayer is a problem or not.

Please summarized what you think is the status as of today?

If we do not need to change the voltage then my wiring on my serial number USA1 system is complete now.
I have just decide to call the system I built out of the homework2 PCB, the PMD assembly serial number USA1 the " USA1 Mocking Krake"

[image]

It is not (yet) mounted to a board and so is still fragile when probing.

nk25719, 2024-10-12:
**DFPlayer Test Result (As of 2024-10-12)**

After testing the DFPlayer with 3.3V, it is functioning properly. We can proceed without making any changes to the setup.

Closing.

ForrestErickson, 2024-10-13:
@nk25719 
The requirement from the DFPlayer web site shows the allowed VCC range as 3.2 - 5V.  
[image]

That is why we can close this issue.

closed 2024-10-13 by ForrestErickson