PubInv/PMD_-pseudo-medical-device-aka-Homework2- #50: Hardware Bug - PCB REV 20240421, J4 (LCD) HAS 3V3 PWR and GDN Reversed Relative to LCD

ForrestErickson, 2024-09-12:
**Describe the Failure / Bug**
_A clear and concise description of what the failure / bug is._
A stright through connection from the PMD to an I2C LCD module crosses power and ground.

_Short Summary of the expected behavior:_
I expected that I could connect with my four wire cables.
_Short Summary of the buggy behavior:_
When connected the USB power to the DEV KIT collapsed (hope it did not damage anything.

**Device Under Test, aka DUT**
Steps to reproduce the behavior:  
1. Model Number:   PMD 
2. Serial Number(s):   SN: USA1
3. Location where failure / bug occurred.  Maryville work bench.
4. Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.)  Lee connected an I2C LCD module using cables from here: https://www.amazon.com/dp/B074C6FV36?ref=ppx_yo2ov_dt_b_fed_asin_title
5. Extent of Failure. All LEDs went off.
** Is failure intermittent?  YES removing the cable and the PMD boots normally
** Is failure fixed by power cycling?  No.
** Is failure permanent?  If the intent is to wire straight to an LCD, it is a design error

**Hardware description
Model:  PMD
Serial Number:  USA1
Other:  I2C LCD

Firmware TBD  Not yet applicable.

ForrestErickson, 2024-09-12:
## Counter Measure By Rewire Cable
The "dupont" connectors are easy to rewire. A small screw driver lifts the lock releasing the crimped terminal.
On one end, swap the Red and Black wires.
Connect the PMD to the I2C LCD such that the SLC and SDA are straight across and the power and gnd are swapped.

nk25719, 2024-10-12:
## **After testing PCB REV 20240421, J4 (LCD) HAS 3V3 PWR and GDN are Correctly assembled Relative to LCD** 

Closing...

nk25719, 2024-10-12:
This issue addressed the workaround for the wires used to connect the LCD to the connector.

solved and **Closing.**