PubInv/PMD_-pseudo-medical-device-aka-Homework2- #52: ESP32 ID for every board

nk25719, 2024-09-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
when we connect more than one esp32 at once we need to have an ID identification to know which module we are connecting to

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
_Identify the unique MAC address for each ESP32 module_

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
__SOLUTION:_ upload a code with a unique ID for each module_ 
code example:
```
#define DEVICE_ID "ESP32-1"  // Change this for each device

void setup() {
  Serial.begin(115200);
  Serial.print("Device ID: ");
  Serial.println(DEVICE_ID);
}

void loop() {
  // Do nothing
}
```

**Additional context**
Add any other context or screenshots about the feature request here.
_This way, we can identify each ESP32 by its custom DEVICE_ID and, be able to distinguish between multiple ESP32 devices when connected._

nk25719, 2024-09-12:
When connecting different ESP32 modules and uploading the firmware to display the MAC address, each of the ESP32 modules displayed the same MAC address. 

# CODE

```
#include "WiFi.h"

void setup() {
  Serial.begin(115200);
  // Get the MAC address
  Serial.print("ESP32 MAC Address: ");
  Serial.println(WiFi.macAddress());
}

void loop() {
  // Do nothing
}
```

# OUTPUT:

[image]

cc: @ForrestErickson

nk25719, 2024-09-16:
Similar output after using the code
WiFiAccessPointWithMAC.ino
on : https://github.com/ForrestErickson/WiFiAccessPointWithMAC/blob/main/WiFiAccessPointWithMAC.ino 

[image]

>ESP32 module A

[image]

>ESP32 module B

nk25719, 2024-09-16:
trying: 

```

#include <WiFi.h>
#include <esp_wifi.h>

void readMacAddress(){
  uint8_t baseMac[6];
  esp_err_t ret = esp_wifi_get_mac(WIFI_IF_STA, baseMac);
  if (ret == ESP_OK) {
    Serial.printf("%02x:%02x:%02x:%02x:%02x:%02x\n",
                  baseMac[0], baseMac[1], baseMac[2],
                  baseMac[3], baseMac[4], baseMac[5]);
  } else {
    Serial.println("Failed to read MAC address");
  }
}

void setup(){
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.STA.begin();

  Serial.print("[DEFAULT] ESP32 Board MAC Address: ");
  readMacAddress();
}
 
void loop(){

}
```

## Result 

[image]

>Mac Address: cc:db:a7:30:09:54

ForrestErickson, 2024-09-16:
Lee copied the code directly above.

It would not compile till line was commented out.
[image]

It then compiled and ran and returned the exact same MAC address as returned by the program Lee wrote.
[image]

This code returns the MAC address (which should be unique) of the ESP kit.

ForrestErickson, 2024-09-16:
Closing because this method found an ESP21 ID.

ForrestErickson, 2024-09-16:
@nk25719 
I have closed this because as written this issue requirement was "when we connect more than one esp32 at once we need to have an ID identification "

It still remains for you to use this MAC address in the KRAKE code and in the PMD code.

I will warn you the code you developed has the address in a format which may be very difficult use.  It is not in one variable either as C character array or a C++ String.

As you try to put this into KRAKE and PMD I can help.

Lee

closed 2024-09-16 by ForrestErickson

nk25719, 2024-09-17:
> <nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/52/2354221980@
> github.com>

ok, I understand,

I am surprised and not able to find the reason why one code pulled the Mac
Address and the second did not,
both worked and pulled the mac address of your ESP3232 modules, right?

Nagham.