PubInv/PMD_-pseudo-medical-device-aka-Homework2- #54: Hardware Bug - LED D8 is ON while No Power Connection on Barrel Jack J1

nk25719, 2024-09-16:
**Describe the Failure / Bug**
A clear and concise description of what the failure / bug is.
Short Summary of the expected behavior:
Short Summary of the buggy behavior:

**Device Under Test, aka DUT**
Steps to reproduce the behavior:  
1. Model Number:   
2. Serial Number(s):   
3. Location where failure / bug occurred
4. Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.)
5. Extent of Failure
** Is failure intermittent?
** Is failure fixed by power cycling?
** Is failure perminant?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
_LED D8 turnung on when the PCBa is connected to power through Barrel Jack J1_

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware description
Model:
Serial Number:
Other:

**Test Equipment
Model:
Serial Number:

**Additional context**
Add any other context about the problem here.

nk25719, 2024-09-17:
[image]

ForrestErickson, 2024-09-17:
Please review the now closed issue: #47 
https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/47

nk25719, 2024-10-27:
Reviewed and solved in issue #47  

cc: @ForrestErickson 

Closing.

closed 2024-10-27 by nk25719