PubInv/PMD_-pseudo-medical-device-aka-Homework2- #56: Add Serial Port and LCD Splash Screen

ForrestErickson, 2024-09-23:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
I am frustrated when I do not know the name and version of the firmware in an embedded system.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Add in the setup() some code to report the desired information on serial port. Here is an example
[image]

Add in the setup() some code to report the desired information on the LCD

ForrestErickson, 2024-09-23:
Here is a modified file with some examples.
This code also reports the date and time of the compile.

[PMDtoKRake_KrakeV2.zip](https://github.com/user-attachments/files/17091891/PMDtoKRake_KrakeV2.zip)

ForrestErickson, 2024-09-28:
Here is the 28 September work.
[PMDtoKRake_KrakeV220240928_1129.zip](https://github.com/user-attachments/files/17174913/PMDtoKRake_KrakeV220240928_1129.zip)

nk25719, 2024-09-29:
> <nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/56/2380709765@
> github.com>

Adding changes to krake's firmware...

ForrestErickson, 2024-09-30:
Note in the "[DFPlayerTD5580ATest.zip](https://github.com/user-attachments/files/17182320/DFPlayerTD5580ATest.zip)" file the serial splash message.

nk25719, 2024-10-12:
```
 Serial.println("===================================");
  Serial.println(DEVICE_UNDER_TEST);
  Serial.print(PROG_NAME);
  Serial.println(VERSION);
  Serial.print("Compiled at: ");
  Serial.println(F(__DATE__ " " __TIME__) ); //compile date that is used for a unique identifier
  Serial.println("===================================");
  Serial.println();

```

Closing.

nk25719, 2024-11-22:
[image]

closed 2024-11-22 by nk25719