PubInv/PMD_-pseudo-medical-device-aka-Homework2- #66: Please see Krake issue number 60 for key cap

ForrestErickson, 2024-10-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Please see Krake issue number 60
https://github.com/PubInv/krake/issues/60

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

nk25719, 2024-11-02:
cc: @samoyin

nk25719, 2024-11-22:
[image]

cc: @ForrestErickson @samoyin

closed 2024-11-22 by nk25719