PubInv/PMD_-pseudo-medical-device-aka-Homework2- #68: Hardware Bug_Incompatible (x) width of the front panel with the box of the enclosure.

samoyin, 2024-10-27:
**Describe the Failure / Bug**
A clear and concise description of what the failure/bug is: The box is not shutting close with the front panel, The x width of the front panel = 62.2mm and the internal x width of the box = 60.8mm.
Short Summary of the expected behavior: A closed enclosure
Short Summary of the buggy behavior: 
[image]

samoyin, 2024-10-27:
[image]
Refinement of measurements to add allowance between the front panel and the box of the enclosure.
path: EnclosureREVc.scad

closed 2024-10-27 by samoyin

nk25719, 2024-10-27:
CC: @ForrestErickson