PubInv/PMD_-pseudo-medical-device-aka-Homework2- #70: LED Seperator

nk25719, 2024-10-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
No tool is available to support LED assembly in a way that ensures they are positioned at a consistent and stable height. To allow all LEDs to be at the same level, high enough to be visible from the top of the front panel, while also maintaining a uniform distance between each LED.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
3D Model Seperator 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

samoyin, 2024-11-02:
Here is a recreation of the GPAD LED standoff using OpenSCAD.

[image]
[image]
These were the measurements that I used to recreate the LED separators. 
Below the measurements is a picture of how the LEDs should look. 

[image]
[image]

nk25719, 2024-11-02:
After 3D printing the part, the diameter of the side holes can increase to double the diameter size that it is.

nk25719, 2024-11-22:
please consider the changes done to the Seperator in the file below, 

[5mm_LED_Separator.zip](https://github.com/user-attachments/files/17866768/5mm_LED_Separator.zip)

note: the hight of theseperator will be effected by the hight of the Enclosure. 

cc: @samoyin

closed 2024-11-22 by nk25719