PubInv/PMD_-pseudo-medical-device-aka-Homework2- #72: Demonstrate the DailyStruggleButton Library for Switch Debouncing, Multi Press and Long Press.

ForrestErickson, 2024-11-08:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
I have made a sketch to demonstrate the library for debouncing a button and more which runs on the Homework2 assembly.
See:
https://github.com/ForrestErickson/ESP32_DailyStruggle-

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Please evaluate this sketch to know of the existence of this library and working code on this hardware.

In addition to debouncing the library can detect multipress (two quick presses) and a long press.

nk25719, 2024-11-22:
[image]

nk25719, 2024-11-22:
[image]

Press the mute button to mute/unmute.  
In case of a stuck button, the button event will not be recognized.