PubInv/PMD_-pseudo-medical-device-aka-Homework2- #74: Elevate LEDs on Homework2 with the GPAD LED Standoff

ForrestErickson, 2024-11-11:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
I want to see how the Homework2 assembly will fit in an enclosure if I use the LED standoff designed for the GPAD enclosure

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Into KiCad PCB foot print for at least one LED, import into a least one LED the GPAD LED Standoff. Spacer file attached below.

Make a STEP file export of the Printed Circuit Assembly

## The GPAD LED Spacer
This file was made by FreeCAD. In the Part work bench the imorted STL file was converted to a mesh and the exported as a STEP file.
[U_Box_V105_GPAD_LED_Standoff_single.zip](https://github.com/user-attachments/files/17693625/U_Box_V105_GPAD_LED_Standoff_single.zip)

ForrestErickson, 2024-11-11:
The spacer added to an LED
[image]

Two LEDs with spacers on the PWA.
[image]

ForrestErickson, 2024-11-11:
Unfortunately, in C:\Users\Public\Downloads\Arduino\PubInv\Nagham\Lee_KiCad-esp32-6leds-2switches-1pj-circuit\KiCad-esp32-6leds-2switches-1pj-circuit\Enclosure_and_Workbench
 I cannot find a file type I can import into FreeCAD.

C:\Users\Public\Downloads\Arduino\PubInv\Nagham\Lee_KiCad-esp32-6leds-2switches-1pj-circuit\KiCad-esp32-6leds-2switches-1pj-circuit\Enclosure_and_Workbench

I can import the OpenSCAD file in FreeCAD but the resulting thing is not parts as reflected in OpenSCAD with a Top and enclosure.
[image]

ForrestErickson, 2024-11-11:
I was able to save the enclosure file as a FreeCAD type and then bring it into the A2Plus assembly for a trial fit like this. 

[image]

It appears to me that there are errors locating all of the LEDs and the Three Terminal Regulator
[image]

nk25719, 2024-11-22:
@samoyin Could we edit the Gpad seperator original design to fit the homework 2 enclosure? 

cc: @ForrestErickson