PubInv/PMD_-pseudo-medical-device-aka-Homework2- #77: Hardware Bug - Fix Rev C

nk25719, 2024-11-22:
**Describe the Failure / Bug**
A clear and concise description of what the failure / bug is.
Short Summary of the expected behavior:
Short Summary of the buggy behavior:

**Device Under Test, aka DUT**
Steps to reproduce the behavior:  
1. Model Number:   EnclosureREVB
2. Serial Number(s):   -
3. Location where failure / bug occurred: USA and LEBANON
4. Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.)
5. Extent of Failure
** Is failure intermittent? no
** Is failure fixed by power cycling? no
** Is failure perminant? yes

**To Reproduce**
Steps to reproduce the behavior:
1. Go to EnclosureREVb and preview 

**Expected behavior**
A clear and concise description of what you expected to happen.
The esp32 at level with the Front Panel 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.
20 mm is the hight of the total PCBa (encluding the ESP32 module kit), 
We need to edit the hight of th enclosure 
and edit the height of the bosses, the seperators and the keycaps

nk25719, 2024-11-22:
Describe the Failure / Bug
A clear and concise description of what the failure / bug is.
Short Summary of the expected behavior: There is a miss centering in the power LED (D1) hole. 
 [image]

And a miss centering in the ESP32 kit hole. 
[image]

Device Under Test, aka DUT
Steps to reproduce the behavior:

Model Number: EnclosureREVC
Serial Number(s): -
Location where failure / bug occurred: USA and LEBANON
Hardware configuration. (How connected, any accessories, what else it takes to replicate the system in which the failure occurred.)
Extent of Failure
** Is failure intermittent? no
** Is failure fixed by power cycling? no
** Is failure perminant? yes

Additional context
Add any other context about the problem here.
Code:
[EnclosureREVc (nagham editted).zip](https://github.com/user-attachments/files/17873937/EnclosureREVc.nagham.editted.zip)

ForrestErickson, 2024-11-22:
Error on PCB makes assembling the mounting screws into the enclosure hard / impossible.
The root cause it in the PCB and must be corrected there.

[image]