PubInv/PMD_-pseudo-medical-device-aka-Homework2- #86: Hardware, Adding Pulse Heart Rate Sensor, d157

nk25719, 2025-05-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
We need Sensors to be added to this PMD

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Adding Pulse Sensor, Heart Rate Sensor, d157

**Additional context**
Add any other context or screenshots about the feature request here.

1- Connect an external wire to GPIO #36
[image]

2- Solder the Signal pin, PWR pin and GND pin of the sensor to terminals
[image]
[image]
[image]

ForrestErickson, 2025-05-03:
Here is an example using the sensor and an ESP32.
https://microcontrollerslab.com/pulse-sensor-esp32-tutorial/

nk25719, 2025-05-06:
## Analog reading ~ 1779 when the SPO2 sensor is not connected

[image]

## Analog reading ~ 2141 + when the SPO2 sensor is connected

[image]

ForrestErickson, 2025-05-06:
Nagham,
Make a user adjustment with a potentiometer of the threshold.
Use the wiper to an analog input pins ADC1_CH4 (or ADC1_CH5).
The the fully counter clock wise pin should go to GND and the fully clock wise should go to 3V3.

Read more about using the ADC from this RNT.
https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/

The user would adjust the potentiometer and watch the heart beat detection LED.